SunOner/HID_Arduino #7: Mouse unresponsive when connect to the host shield

Jotalz, 2024-03-14:
Hello, SunOner,when i connect the mouse to the host shield,nothing happens, I'm using logi G102,It's like there's no electricity and there's no light effect,of course it cannot be responded to.
In the debug output of the Arduino IDE, only the moment I connect the mouse USB to the host shield produces output, is as follows：
`BM Init`
`Addr:01`
`NC:01`
`HID_PROTOCOL_MOUSE`
`bNumEP:01`
`D1`
I have done all the previous steps, there should be no problem.
Is it a problem of insufficient power supply?

Jotalz, 2024-03-14:
I can control it perfectly using G PRO WIRELESS connection. It seems this is a mouse compatibility issue. Can it be solved? I'd rather use that mouse to control

Jotalz, 2024-03-29:
Thank you for creating a project for Logitech G102/G502, it's really great!

closed 2024-03-29 by Jotalz

Hugezom, 2024-05-25:
Hi Jotalz, i have the same problem as you. My G102 has no electricity and light effect even with the G series project. I would appreciate if you provide any information about how to make it work

Jotalz, 2024-05-26:
Read the author's [note](https://github.com/SunOner/usb-host-shield-mouse_for_ai_aimbot), he created a repo exclusive to Logitech G series for us!Other steps are consistent with the main.

FoxtrotBravoSix, 2024-05-28:
> Read the author's [note](https://github.com/SunOner/usb-host-shield-mouse_for_ai_aimbot), he created a repo exclusive to Logitech G series for us!Other steps are consistent with the main.

Where is USBHostShieldMouse.ino sketch he talking about? There is only hidmousereport.ino