SunOner/HID_Arduino #8: is it necessary to use usb host shield

Hugezom, 2024-05-25:
I'm curious about the role usb host shield plays here. I mean what is the difference between using a USB Host Shield to make the Leonardo the sole mouse input, and using the Leonardo and mouse separately? If there are some reasons like Anti-Cheat programs would check the number of input devices? 
Any infomation would be helpful

ysy134, 2026-09-02:
I think one purpose of using the expansion board is to enable mouse passthrough – merging real mouse operations with computer‑generated simulated inputs, then sending the combined data to the RP2040. The RP2040 emulates a real mouse to the PC, thereby bypassing detection that often flags software‑emulated mouse data.

NANOBYTECOMPUTERS, 2026-09-02:
The host sheild translates the mouse output from the synthetic generated mouse deltas from the program, as well as the real mouse and makes it appear to be a single mouse. The u s b host shield is simply to host the mouse so that it does not appear to be too separate devices plugged in. I have taken it a step further with teensy and made it to where you can change the vendor and product I d so that it's not a solitary u s b. Device they are looking for lessening the likelihood of detection