SunOner/HID_Arduino #11: не работает в игре warafce

swolvla, 2024-11-01:
в меню мышь двигается, но кнопки не работают, а в игре она вообще не реагирует даже на движение. вне игры мышь работает, но постоянно прокручивается колесо мыши либо вверх либо вниз, зависит от того как настроить переменные в MYMOUSEINFO

dnlksnvv, 2024-12-10:
> в меню мышь двигается, но кнопки не работают, а в игре она вообще не реагирует даже на движение. вне игры мышь работает, но постоянно прокручивается колесо мыши либо вверх либо вниз, зависит от того как настроить переменные в MYMOUSEINFO

Такая же проблема. Перепробовал множество вариантов реализации движения мышью в WF. Нашел решение проблемы?

SunOner, 2024-12-10:
> > в меню мышь двигается, но кнопки не работают, а в игре она вообще не реагирует даже на движение. вне игры мышь работает, но постоянно прокручивается колесо мыши либо вверх либо вниз, зависит от того как настроить переменные в MYMOUSEINFO
> 
> Такая же проблема. Перепробовал множество вариантов реализации движения мышью в WF - всё бесполезно. Нашел решение проблемы?

Без "спуфа" ардуино ничего работать не будет. На Discord сервере в канале "error-and-tips" можно найти тред "Change Arduino name". После всех внесённых изменений , всё будет работать.

closed 2024-12-10 by SunOner